forkineye/ESPixelStick #812: FileManager doesn't read file after initial load

Terrevue, 2024-12-12:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
FW Version. 4.0-ci12244189758
Build Date. Dec 9 2024 - 21:01:15
Architecture. ESP32
Board Name. esp32_devkitc
Flash Size. 4194304
Flash ID. 4FE4BBBBC2C

**Describe the bug**
See log below.  
1.) Can't find/setup SD Card
2.) Finds SD card
3.) Finds files
4.) Can't find "fseqfilelist.json" (confirmed that it is there)
5.) Can't load files

[FileMgr] No SD card installed
[FileMgr] SD 'Card' setup failed.
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Set SD speed to 25Mhz
[FileMgr] SD Card Size: 64088MB
[FileMgr] SD File: 'Defying Gravity - 20241209b.fseq'
[...]

[FileMgr] Configuration File: '/input_config.json' loaded.
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] ERROR: Cannot find 'wificonfig.json' for reading. File does not exist.
[FileMgr] SD file: 'wificonfig.json' not found.
[FileMgr] Configuration File: '/admininfo.json' saved 171 bytes.
[Effects] Start EffectsTask
[FileMgr] Configuration File: '/config.json' saved 513 bytes.
[WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[WiFiDrv] Using DHCP
[WiFiDrv] Connecting to 'xxxx' as esps-001
[WiFiDrv] WiFi Failed to connect using Configured Credentials
[WiFiDrv] WiFi Entering State: Connecting As AP
[WiFiDrv] WiFi SOFTAP:       ssid: 'esps-001
[WiFiDrv] WiFi SOFTAP: IP Address: '0.0.0.0
[WiFiDrv] WiFi Entering State: Connected To STA
[WiFiDrv] Connected to STA with IP: 192.168.4.1
[   DDP] Listening on port 4048
[WebMgr] Web server listening on port 80
[FileMgr] Starting FTP server.
[  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[ 58202][E][vfs_api.cpp:105] open(): /littlefs/admininfo.json.gz does not exist, no permits for creation
[FileMgr] ERROR: Cannot find 'fseqfilelist.json' for reading. File does not exist.
[WebMgr] ERROR: Could not open List of Fseq files for reading
[FileMgr] ERROR: Could not Open SD file list.
[FileMgr] CloseSdFile::ERROR::Invalid File Handle: 0
[116444][E][vfs_api.cpp:105] open(): /littlefs/admininfo.json.gz does not exist, no permits for creation
[FileMgr] ERROR: Cannot find 'fseqfilelist.json' for reading. File does not exist.
[WebMgr] ERROR: Could not open List of Fseq files for reading
[FileMgr] ERROR: Could not Open SD file list.
[FileMgr] CloseSdFile::ERROR::Invalid File Handle: 0

MartinMueller2003, 2024-12-12:
ESP File system supports a maximum of 32 GB SD cards, you have a 64 GB card installed. I suggest using a super high speed 8GB or 16 GB card.

Terrevue, 2024-12-12:
Does this have something to do with which device is used during flashing?  On my QuinLED32 AE+ devices, the files read fine from a 64gb card:
[image]

On the other device, I used a esp32_devkitc, but since they are both essentially the same ESP32 chip, I wouldn't think it would matter

MartinMueller2003, 2024-12-19:
It depends. However, is your other device actually a devkitC? For most systems, I am not seeing such issues. The boot routine tries to bring up the SD card twice (one retry after first failure) which is why you see the initial failure.

MartinMueller2003, 2024-12-28:
I am still not having an issue with this on my Dig Quad or Tetra2go boards.

closed 2025-03-18 by MartinMueller2003