forkineye/ESPixelStick #816: main input: DDP + secondary input: effects results in flickering

dutch-anykey, 2024-12-20:
**ESPixelStick Firmware Version**
FW Version 4.0-ci12223533891
Build Date Dec 8 2024 - 16:47:59
Architecture ESP32
Board Name esp32_wasatch
Flash Size 8388608
Flash IDE4C12D65B7A0

**Binary release or compiled yourself?**
compiled myself 

Still playing around with my WT32 with wasatch build.
All running fine but noticed flickering on outputs when enabling effects + effect flash settings. I wondered if it was the same problem as reported by other users. (Flickering on output) But it is really the effect flash settings.

For the problem. I assumed that when an effect is running and when DDP traffic arrives that the effect would stop. But instead it looks like both are trying to output to the ports and I get gardbled/flickering led's on the output. 
This looks like a bug. 
I'm outputting DDP traffic directly from Xlights, not from FPP player. 

br
Maurice

MartinMueller2003, 2024-12-20:
The bad flashing has been resolved on my fork. This looks like the flicker function is not looking at the "channel enabled" flag

dutch-anykey, 2024-12-21:
Just to get my confusion out of the way. With flicker function you mean effect flash setting or the effect itself?
What I'm seeing is really seeing partly the effect (i.e. I selected Marquee) and the DDP input (sequence playing from Xlights)
Espixelstick is outputting DDP and effect at the same time.

MartinMueller2003, 2024-12-28:
A fix for this is in a PR and on the web flasher are "Experimental"

closed 2024-12-28 by MartinMueller2003