forkineye/ESPixelStick #817: cache problem after updating via efu

dutch-anykey, 2024-12-25:
Merry Xmas all :)

After updating to the last CI build I noticed that the version number wasn't updated in the gui. I updated via an EFU image created via https://espixelstickwebflasher.from-ct.com:5000/ 
Version prior to the update was CI build 4.0-ci1224....

**FW Version 4.0-ci12414601140**
Build Date Dec 19 2024 - 14:29:07
Architecture ESP32Board 
Name esp32_wasatch
Flash Size8388608

Looks like a bug is back that was recently solved? After clearing the cache in my chrome browser it showed the version correctly.

kind regards,
Maurice

dutch-anykey, 2024-12-25:
Weird...

For test I did a factory reset and noticed that there is no problem with cache anymore. all data was gone as it should be in the chrome browser

MartinMueller2003, 2024-12-27:
It really depends on the ending version. FYI: I just made more changes to the EFU. Added user feedback on both success and failure.

dutch-anykey, 2024-12-30:
Thanks Martin.

closed 2024-12-30 by dutch-anykey