forkineye/ESPixelStick #820: build problems related to: ArduinoJson::V730PB22

dutch-anykey, 2024-12-30:
**ESPixelStick Firmware Version**
latest build 

Trying to build from the latest ci but get errors related to ArduinoJson::V730PB22

part of the error log from VScode:

[{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V730PB22::detail::enable_if<false, int>'",
	"startLineNumber": 192,
	"startColumn": 74,
	"endLineNumber": 192,
	"endColumn": 74
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V730PB22::detail::enable_if<false, int>'",
	"startLineNumber": 203,
	"startColumn": 40,
	"endLineNumber": 203,
	"endColumn": 40
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V730PB22::detail::enable_if<false, int>'",
	"startLineNumber": 212,
	"startColumn": 74,
	"endLineNumber": 212,
	"endColumn": 74
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V730PB22::detail::enable_if<false, int>'",
	"startLineNumber": 223,
	"startColumn": 40,
	"endLineNumber": 223,
	"endColumn": 40
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V730PB22::detail::enable_if<false, int>'",
	"startLineNumber": 232,
	"startColumn": 71,
	"endLineNumber": 232,
	"endColumn": 71
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V730PB22::detail::enable_if<false, int>'",
	"startLineNumber": 246,
	"startColumn": 76,
	"endLineNumber": 246,
	"endColumn": 76
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/.pio/libdeps/esp32_devkitc/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'convertToJson(const char [], ArduinoJson::V730PB22::JsonVariant&)'",
	"startLineNumber": 30,
	"startColumn": 18,
	"endLineNumber": 30,
	"endColumn": 18
[... 147 lines cut ...]
	"severity": 8,
	"message": "assignment of read-only location '*(((const char*)(& CN_long)) + ((sizetype)JsonData.ArduinoJson::V730PB22::JsonObject::operator bool()))'",
	"startLineNumber": 76,
	"startColumn": 26,
	"endLineNumber": 76,
	"endColumn": 26
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/src/input/externalInput.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "assignment of read-only location '*(((const char*)(& CN_id)) + ((sizetype)JsonData.ArduinoJson::V730PB22::JsonObject::operator bool()))'",
	"startLineNumber": 89,
	"startColumn": 23,
	"endLineNumber": 89,
	"endColumn": 23
},{
	"resource": "/e:/espixelstick_build_folder/ESPixelStick-main30-12-24/ESPixelStick-main/src/input/externalInput.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "assignment of read-only location '*(((const char*)(& CN_state)) + ((sizetype)JsonData.ArduinoJson::V730PB22::JsonObject::operator bool()))'",
	"startLineNumber": 90,
	"startColumn": 47,
	"endLineNumber": 90,
	"endColumn": 47
}]

MartinMueller2003, 2024-12-30:
Looking at it now. I am testing with 7.2.1 and somehow git actions are using 7.3.0 and that has issues.

MartinMueller2003, 2024-12-30:
Just pushed a version that moves back to arduinojson 2.7.1 until I can figure out what is wrong.

MartinMueller2003, 2025-01-23:
I have made some significant changes and we are now moved forward to 2.7.3

closed 2025-01-23 by MartinMueller2003