forkineye/ESPixelStick #823: several observations latest (local) build

dutch-anykey, 2025-01-08:
Hi Martin,

I'm testing your latest version by building it locally. (Not the latest CI build. espixelstickwebflasher seems down again)

FW Version 4.x-dev (downloaded zip file from 8 jan 2025 martin mueller)
Build Date Jan 8 2025 - 07:21:13
Architecture ESP32Board 
Name esp32_devkitc
Flash Size 4194304
Flash ID9C0B8D0A8AB4

Some observations:

1 - Maybe a bug? I can enable FPP remote although there is no SD card connected.
Not bothering me, but in the past "fpp remote" on the secondary input was not selectable if SD card was absent.

After installing SD card:

2 - automatic switching between primary and secondary input seems to be working now (No concurrent outputs anymore) However I noticed a few reboots when switching inputs.
Setup is playing live DDP data from Xlights and configured effects on secondary input. When stopping the sequence in Xlights espixelstick switches to the configured effect. And also the effect stops when starting the seq. again from xlights. Tried switching a few times but after a few times switching back and forward I noticed a few reboots.

below the output from the serial console:

 

[ 96883][E][vfs_api.cpp:105] open(): /littlefs/config.json.gz does not exist, no permits for creation
[109949][E][vfs_api.cpp:105] open(): /littlefs/admininfo.json.gz does not exist, no permits for creation
[110085][E][vfs_api.cpp:105] open(): /littlefs/input_config.json.gz does not exist, no permits for creation
[110227][E][vfs_api.cpp:105] open(): /littlefs/output_config.json.gz does not exist, no permits for creation
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4018d7d7  PS      : 0x00060d30  A0      : 0x8018d9c7  A1      : 0x3ffaf2b0  
A2      : 0x00000000  A3      : 0x3ffdfe60  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffdaf8c  A7      : 0x00000000  A8      : 0x80084d0d  A9      : 0x3ffaf2b0  
A10     : 0x3ffdfa70  A11     : 0x00001800  A12     : 0x3ffdfa70  A13     : 0x00000001
A14     : 0x00060d20  A15     : 0x00000001  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008b1a0  LEND    : 0x4008b1b6  LCOUNT  : 0xffffffff

Backtrace: 0x4018d7d4:0x3ffaf2b0 0x4018d9c4:0x3ffaf2f0 0x4018da15:0x3ffaf310 0x4013921d:0x3ffaf340 0x4013dffe:0x3ffaf3b0 0x4014204a:0x3ffaf3e0 0x40132eb9:0x3ffaf400

ELF file SHA256: 673df002762e6a7d

E (5934) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (539) esp_core_dump_flash: No core dum   ѥѥ   found!
E (539) esp_core_dump_flash: No core dump partition found!

[   ESP] ESPixelStick v4.x-dev (Jan  8 2025 - 07:21:13)
[   ESP] v4.4.7-dirty
[   ESP] CPU Frequency: 240MHz
[   ESP] APB Frequency: 80MHz
[FileMgr] Flash file system initialized. Used = 233472 out of 327680

3 - upload of sequence via espixelstick gui works, but via fpp i get error 100 each time:

[image]

GeorgeLJr, 2025-01-26:
Good afternoon. Wondering if anything came about with the error code 100? I'm having the same issue when I try to FPP connect and upload sequence Error uploading file, Response code 100. I was suggested to update back to a September version but I'd have to learn how to do that as I'm new to this hobby. I was hoping there might be a fix for it or maybe I'm doing something wrong. I'm running FW Version4.0-ci12520549028Build DateDec 27 2024 - 20:51:18 right now. Thanks

dutch-anykey, 2025-01-26:
@GeorgeLJr 

It's an odd one. I had the problem and later that day I couldn't reproduce it anymore. I haven't been able to figure out what I did change in-between.
Some things that I would try:

- check that SD card is working (perform an upload via the espixelstick gui)
- if there are problem format the SD card on your laptop/pc
- try a different fseq file (should be v2 uncompressed)
- factory reset (via admin page)

If you want to start over and flash it again I would go via https://espixelstickwebflasher.from-ct.com:5000/
There are different versions from the pull down menu.

kind regards,
Maurice

GeorgeLJr, 2025-01-28:
Thank you for the response. This definitely is not user friendly as I can't get anything to work. I'm sure it's me. I'll keep trying for a few and If I can't do it, I'll just toss it and try a different controller. Thanks anyway!

GeorgeLJr, 2025-01-28:
thanks for the response. I went back to the September version and now I'm able to load the sequence with no error code but now I can't save settings when I change it to FPP remote...

GeorgeLJr, 2025-01-28:
I flashed the card again with the December 8th version and it works!! So guess something isn't quite right in the newest version for me. Thanks again for the help, although I didn't understand more than half of it, it still got me where I need to be so thanks again!

MartinMueller2003, 2025-02-03:
The xLights team and I have been working on this. There are parts of the issue on both sides. We have now added zip file support and I have changed how the file data buffer is being managed. The buffer management changes have significantly reduced the error 100 and (imo) eliminated the error 0 issues. Using zip files improves overall transfer times but requires a reboot after uploading the files. Also, the zip support is only available on the ESP32 versions.

Other than that, the latest build on my fork should resolve most of these issues.

dutch-anykey, 2025-02-04:
Thanks Martin !

Looks like a lot of work has been done. I have build and upload your version in VScode. First tests are looking good, also the Xlights improvements. I will test more later this week.

kind regards,
Maurice

GeorgeLJr, 2025-02-09:
Finally getting around to updating. Last time I updated to an older version and it worked great. Well, not for long. Went back to code 100 but I updated everything this morning to the current and uploaded a much bigger sequence and it took a bit but it worked perfectly! Hopefully it keeps working! Thank you so much!

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

MartinMueller2003, 2025-02-18:
We did more work on this over the last few days. xLights has a PR open for changes on their side and I have a PR open for changes on the ESPS. Testing over the last days has shown good download rates from xLights to ESPS using ESP32 with zip support enabled.

MartinMueller2003, 2025-02-18:
Still working on allowing FPP remote to override local fseq file play.

MartinMueller2003, 2025-03-18:
This is working now. FPP Override and then play local file after current file finishes.

closed 2025-03-18 by MartinMueller2003