forkineye/ESPixelStick #826: play a default fseq file when no FPP sync traffic is coming

dutch-anykey, 2025-01-11:
Feature request:

I built 20 hardware controllers based on the devc esp32 kit with SD connected. These are going to be used on led-suits. (We as a friends group participate in a local carnaval light parade) 
The plan is to use them in “FPP remote” mode so each controller plays its own fseq from SD card and sync via wifi.
However I anticipate that sometimes people walk out of range of the wifi coverage. For that it would be great if it starts playing a “default” fseq file on the SD card. 

Feature request:
Is it possible to detect if no FPP sync messages are coming in for x seconds and then start playing a default fseq file on the sd card?
I.e. primary input will be disabled, secondary input: FPP Remote. Ad an option under FPP remote configuration like “on idle play default.fseq”
And of course when FPP sync starts again to continue playing FPP remote mode. 

Br
Maurice

MartinMueller2003, 2025-02-05:
Yes this should be possible. It is going to take a while and what is really going to happen is:

ESPS receives an FPP sync message and starts playing at the specified file location. 
As long as sync messages are received, then the FPP file will get played.
If FPP sync message stop coming in, the current file will play until completion.
After the current file completes, the system will wait through the blank period (configurable) and then start playing the default file.

dutch-anykey, 2025-02-06:
Hi Martin,

That sounds good. Looking forward. :) 
Thanks.

MartinMueller2003, 2025-02-21:
Added playing a background fseq file when FPP is not sending anything. PR #832