forkineye/ESPixelStick #830: wt32 wasatch choppy performance on outputs  latest build from Martin Meuller

dutch-anykey, 2025-02-10:
Hi Martin,

You probably don't even want to hear about wt32 haha. But I'm going to report my findings anyway. I'm not even sure if this is related to the wt32 board.

So I have one running with CI 2223533891 (dec. 8th 2024) And it is running stable surprisingly. But now I have build the latest version from your  repository (today)

For test I enabled 8 outputs (ws2811 150 leds per channel) and started an effect. it is the marquee effect with 3 colors like below:

[image]
 
I notice that outputs can't keep up. with 2 instead of  3 colors it somehow looks okay.
Second test is to play DDP data from Xlights. From there I also observe that it is skipping and choppy performance.  The uptime counter on the homepage of the ESP is also choppy and skipping.

When I revert back to CI 2223533891 all is okay. Performance is there and the output on the leds are smooth.

Not sure where it changed because I haven't tested the 2 versions that are in-between. If it helps I can do so.
I will also start testing with other non wt32 boards and see if I can find some related problems there.

kind regards,
Maurice

MartinMueller2003, 2025-02-18:
I will take a look.

MartinMueller2003, 2025-03-18:
Just made some changes in PR #837 that should resolve this.

MartinMueller2003, 2025-04-03:
Going to close this as there have been improvements in the output processing.