forkineye/ESPixelStick #842: Rest API to turn on/off relay

onewithhammer, 2025-04-04:
Add a rest API (aka http command) that tells the EPS to turn on or off the relay output.

This could be used to turn on/off the main power supply when not in use.

MartinMueller2003, 2025-04-06:
http://192.168.10.220/relay/1/129
.../relay/RelayId/outputValue 0 - 255

MartinMueller2003, 2025-04-06:
Instead of an intensity value you can also use ON or OFF

kmillecam, 2025-04-10:
@MartinMueller2003, I'm testing this feature and may be doing something wrong.
Can you help me find my way?

1. I installed the "4.0-Experimental" version of the ESPixelStick firmware.
2. Under the "Network Setup" tab, changed the "Power Control Pin" to 15 (which matches the Relay config in WLED. See attached).
3. Saved changes.
4. From a browser, visited http://[Dig Quad address]/relay/ON
5. Got a "Relay Output Not configured" message in my browser window
6. From a browser, visited http://[Dig Quad address]/relay/OFF
7. Got the same "Relay Output Not configured" message

Can you give me a nudge in the right direction?

Thanks,
Kevin

[image]

MartinMueller2003, 2025-04-11:
You missed one teensy tiny parameter
http://[Dig Quad address]/relay/1/ON

You need to include the number of the relay. This number corresponds to the relay on which you have enabled HTTP control (A step you did not mention either).

kmillecam, 2025-04-11:
Thanks for the nudge @MartinMueller2003.

I just noticed that my "Device Setup" screen is different than the one @onewithhammer showed in his video tutorial. I'll attach a capture of the screen I'm seeing. (It's missing the dropdown selectors for Primary and Secondary inputs).

Any idea why I'm seeing a different screen?

[image]

MartinMueller2003, 2025-04-11:
please do a factory reset from the admin page. It looks like the system did not properly initialize after the flash.

joe8012, 2025-04-12:
I'm having the same problem except all of my tabs on the UI are blank, including the network tab.  However, the ESPS is showing connected(ip address) in my routers client list.  Latest experimental version(4/7/2025).  Factory reset did not help.

MartinMueller2003, 2025-04-12:
I just load quinled quad ae+ and tetra2go via the web flasher and they loaded properly.

[image]

MartinMueller2003, 2025-04-12:
noman Found the issue. I had left a debug statement active that forced the data to be sourced from an esp at 192.168.10.220. Only an ESP at that address would work.

I have uploaded an experimental image that does NOT have that restriction.

kmillecam, 2025-04-14:
Thanks Martin.
I'll give it a test run Monday.

kmillecam, 2025-04-14:
@MartinMueller2003, @onewithhammer:

To make this work, I believe I need to use the "Device Setup" screen to configure one of the outputs to be a Relay. (Please correct me if I'm wrong.) I'm no longer seeing the "Relay" option in the dropdown to configure output 5. (See attached.)

[image]

MartinMueller2003, 2025-04-14:
For the Quad SE+ (Which has 8 choices) the relay selection is in output 8. And if you are getting this far, then you definitely are not seeing the reboot loop.

onewithhammer, 2025-04-14:
I don't see a relay port on the AE+ (1-8 ports)

onewithhammer, 2025-04-14:
The AE+ 8 port version of the firmware doesn't support it.  Make sure you flash the AE+ version.

onewithhammer, 2025-04-15:
I tested with the following setup.  The relays turn ON but do not latch.

https://docs.google.com/drawings/d/1d6fShU3XQw6l8Sk7Ky7BlcysrOO1Kgrol1zg4GNDnwI/edit?usp=sharing

MartinMueller2003, 2025-04-15:
New experimental version has been uploaded. Fixes the latching issue and the initial ESP8266 reboot issue.

kmillecam, 2025-04-15:
Is this the right syntax for the relay control?
http://192.168.10.76/relay/1/OFF
My browser is responding with "OK" but the relay is not responding.

[image]
[image]

kmillecam, 2025-04-16:
Thanks Martin.
I'll check in with Tony and follow up.

MartinMueller2003, 2025-04-16:
Hi Kevin

Just looked at your picture again. If you are using mechanical relays, you need to turn off PWM output. If you are using SSRs, then you would use PWM.

onewithhammer, 2025-04-16:
Kevin - I tested my setup and it is working properly.  I have the PWM turned on and the relays work with ON/OFF but not with the threshold values.  I turned PWM off and the threshold values worked properly.  You need to set the GPIO values.  See my screenshot.

http://[Dig Quad address]/relay/1/ON  - Turn on relay #1
http://[Dig Quad address]/relay/1/OFF - Turn off relay #1
http://[Dig Quad address]/relay/4/ON  - Turn on relay #4
http://[Dig Quad address]/relay/4/OFF - Turn off relay #4
http://[Dig Quad address]/relay/1/[1-255] - Use threshold value to determine if relay #1 is on/off
If threshold is set to 127
then 127 is ON, 128 is ON and 126 is OFF

[image]

kmillecam, 2025-04-16:
Thanks again for all your help Martin and Tony.
I was able to figure out the proper configuration for my setup. (Yay!) I'm using a pre-built controller sold with Dr. Zzs Permatrack lighting (see image below). This controller (and the Dig-Quad it's build around) uses what the documentation calls a, "Q1R 5.1v relay trigger pin". Once I figured out that that trigger pin uses GPIO 15, things were almost working. I did have to uncheck the "inverted" checkbox in order to make the "ON" switch turn the power to the PS On and not Off (and vise versa).

[image]

[image]

onewithhammer, 2025-04-16:
Kevin - You may want to turn off PWM since the relay appears to be a mechanical relay.  I had PWM turned on in my video.  It will work but the Threshold value doesn't work properly.