forkineye/ESPixelStick #848: Add KSZ8863 Multi port Ethernet Support

MartinMueller2003, 2025-04-21:
Add support for the multi channel Ethernet device KSZ8863.

noman1228, 2025-04-22:
Possibilities go endless with this guy... Daisy-Chain and add power...

noman1228, 2025-04-27:
pioarduino?

MartinMueller2003, 2025-04-27:
My initial pro0blem seems to have been Win 11 file name length issues. These are resolved by making a config change in Windows to support long file names. Now I get compile issues because the new files do not appear in the the pio include paths and it looks like the build is trying to use the old framework. One step at a time :)

noman1228, 2025-04-27:
I know your pain and have complete faith in your resolve. Just remember, figuring stuff out is usually the driving force for us like-minded. It's not as fun once it works as expected...

MartinMueller2003, 2025-04-27:
But it would be nice get a freebee every once in a while :)

The build is using:

.pio\packages\framework-arduinoespressif32\tools\sdk\esp32\include\lwip\lwip\src\include\lwip\prot\ethernet.h

when it should be using:

.pio\packages\framework-arduinoespressif32-libs\esp32\include\lwip\lwip\src\include\lwip\prot\ethernet.h

noman1228, 2025-04-27:
[esp32]
platform = espressif32 @ 6.10.0
board_build.filesystem = littlefs
board_build.partitions = ESP32_partitions.csv

build_flags =
    ${env.build_flags}
    -D ARDUINO_LOOP_STACK_SIZE=12288
    -D DEFAULT_FTP_SERVER_NETWORK_TYPE_ESP32=6
    -D DEFAULT_STORAGE_TYPE_ESP32=2
    -D SUPPORT_UNZIP=1
    -D CONFIG_DISABLE_HAL_LOCKS=1
    -D CONFIG_RMT_ISR_IRAM_SAFE=yes
    -D DEBUG_FILE_HANDLES=1
    -I.pio/packages/framework-arduinoespressif32-libs/esp32/include

build_unflags =
    ${env.build_unflags}
    -D BOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -mfix-esp32-psram-cache-strategy=memw
    -I.pio/packages/framework-arduinoespressif32/tools/sdk/esp32/include
    -I.pio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip

lib_deps =
    ${env.lib_deps}
    mathieucarbou/AsyncTCP @ ^3.2.14
    mathieucarbou/OneWire @ ^2.3.9
    https://github.com/matmunk/DS18B20

extra_scripts =
    ${env.extra_scripts}
    .scripts/replace_fs.py

noman1228, 2025-04-27:
Obvious guess without being in front of anything remotely approaching a dev PC

noman1228, 2025-04-27:
If you swap to the hybrid non-official versions, the can of worms is immense. RMT will need complete re-write. I've failed a few times and while I know you're absolutely the man for the job, the level of time dedication towards a single feature set is insanity.

PKCubed, 2025-08-07:
Any updates on this? I would really like to see the Arduino framework supporting the KSZ8863.