forkineye/ESPixelStick #855: Pixel stick not displaying  sequences on sd card in the FPP remote dialog

patdelaney, 2025-06-26:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
FW Version
4.0-ci15912260092
Build Date
Jun 26 2025 - 20:54:21

**Hardware Version**
Pixelstick V3 D! mini clone

**Binary release or compiled yourself?**
Latest released from web flasher

**Web Browser (and version)**
Chrome

**Access Point**
no

**Describe the bug**
I uploaded a sequence to the pixelstick, it is displaying in File management as  ppd-demoV2.xlz.. I'm guessing that is xlights zip format?  It's been a while since I played with this.

[image]

When I go to Device setup and configure Secondary input to FPP Remote, All I get in the FESQ File to play dropdown box is Play Remote Sequence

[image]

patdelaney, 2025-06-27:
> That is working as designed. After doing an fpp connect you need to
> manually perform a reboot on the esps. The esps will unpackage those files
> and turn them into fseq files that will show on the list. Did you do the
> reboot?

Ah I did not know that, I had given up and powered it off. I powered it
back up and checked the files were still zipped.   I deleted them and I
resent them via fpp connect it transferred the file.  I issued a reboot
from the admin tab, but the file is still there.  This is a D1 MINI  not a
32

Pat

patdelaney, 2025-06-27:
> As in an ESP8266? xLights is not supposed to send a zip file to an ESP8266
> based device. Unzip is only available on the ESP32

Ah, I see.

patdelaney, 2025-06-27:
What is the API call to pixelstick to get the status

MartinMueller2003, 2025-06-28:
I just saw your response to Alex on slack (I am not in that group). The json record you showed is exactly what alex was looking for. The next bit of information is that he is sending you an uncompressed file and then adding the compressed file extension. This is incorrect. :)

MartinMueller2003, 2025-06-28:
Just updated the experimental version with code to reboot after xLights finishes downloading files. Also added code for the ESP8266 to rename the .xlz file to an fseq file.

patdelaney, 2025-06-28:
I used the webflasher to update octa2go with the dev build and the Wifi
appears to be flaky on the Octa2go
I will attach the logs, but after the initial load, it does connect to
wifi, but when you then when you reboot it will not connect again and it
goes into AP mode.

I flashed it a few times with
[image: image.png]

[11:18:12]���C[FileMgr] No SD card installed
[11:18:12][FileMgr] SD initialization failed - code: 1
[11:18:12]Card reset failed
[11:18:12][FileMgr] Configuration File: '/fseqfilelist' saved 76 bytes.
[11:18:12][FileMgr] Configuration File: '/config.json' loaded.
[11:18:12][FileMgr] Configuration File: '/input_config.json' loaded.
[11:18:12]Stop server!
[11:18:12][FileMgr] Configuration File: '/output_config.json' loaded.
[11:18:12][FileMgr] Configuration File: '/admininfo.json' saved 172 bytes.
[11:18:12] Ftp server waiting for connection on port 21
[11:18:12][FileMgr] Configuration File: '/config.json' saved 530 bytes.
[11:18:13][WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[11:18:13][WiFiDrv] Using DHCP
[11:18:13][WiFiDrv] Connecting to 'paddycakes-2.4g' as Octa2Go
[11:18:29][WiFiDrv] WiFi Entering State: Connected To AP
[11:18:29][WiFiDrv] Connected with IP: 192.168.1.181
[11:18:29][WebMgr] Web server listening on port 80
[11:18:29]Stop server!
[11:18:29][  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[11:18:29] Ftp server waiting for connection on port 21
[11:18:29][  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[11:19:37]ets Jul 29 2019 12:21:46
[11:19:37]
[11:19:37]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[11:19:37]configsip: 0, SPIWP:0xee
[11:19:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[11:19:37]mode:DIO, clock div:1
[11:19:37]load:0x3fff0030,len:1184
[11:19:37]load:0x40078000,len:13232
[11:19:37]load:0x40080400,len:3028
[11:19:37]entry 0x400805e4
[11:19:38]E (465) esp_core_dump_flash: No core du� ���ѥѥ���found!
[11:19:38]E (465) esp_core_dump_flash: No core dump partition found!
[11:19:38]
[11:19:38][   ESP] ESPixelStick v4.0-ci15940051127 (Jun 28 2025 - 03:40:26) on esp32_octa2go
[11:19:38][   ESP] v4.4.7-dirty
[11:19:38][   ESP] CPU Frequency: 240MHz
[11:19:38][   ESP] APB Frequency: 80MHz
[11:19:38][FileMgr] Flash file system initialized. Used = 233472 out of 327680
[11:19:38][FileMgr] Listing directory: /
[11:19:38][FileMgr] 'UpdRecipe.json': 	'784'
[11:19:38][FileMgr] 'admininfo.json': 	'172'
[11:19:38][FileMgr] 'config.json': 	'530'
[11:19:38][FileMgr] 'fseqfilelist': 	'76'
[11:19:38][FileMgr] 'input_config.json': 	'2064'
[11:19:38][FileMgr] 'output_config.json': 	'13503'
[11:19:38][FileMgr] Listing directory: //www
[11:19:38][FileMgr] 'alexa.html.gz': 	'146'
[11:19:38][FileMgr] 'apa102.html.gz': 	'1503'
[11:19:38][FileMgr] 'artnet.html.gz': 	'385'
[11:19:38][FileMgr] 'ddp.html.gz': 	'144'
[11:19:38][FileMgr] 'disabled.html.gz': 	'49'
[11:19:38][FileMgr] 'dmx.html.gz': 	'455'
[11:19:38][FileMgr] 'e1_31.html.gz': 	'460'
[11:19:38][FileMgr] 'effects.html.gz': 	'1051'
[11:19:38][FileMgr] 'esps.css.gz': 	'13424'
[11:19:38][FileMgr] 'esps.js.gz': 	'62704'
[11:19:38][FileMgr] 'favicon.ico': 	'6450'
[11:19:38][FileMgr] 'fpp_remote.html.gz': 	'428'
[11:19:38][FileMgr] 'gece.html.gz': 	'501'
[11:19:39][FileMgr] 'grinch.html.gz': 	'377'
[... 103 lines cut ...]
[11:19:56][FileMgr] 'ucs1903.html.gz': 	'1580'
[11:19:56][FileMgr] 'ucs8903.html.gz': 	'1585'
[11:19:56][FileMgr] 'ws2801.html.gz': 	'1502'
[11:19:56][FileMgr] 'ws2811.html.gz': 	'1582'
[11:19:58][FileMgr] No SD card installed
[11:19:58][FileMgr] SD initialization failed - code: 1
[11:19:58]Card reset failed
[11:19:58][FileMgr] Configuration File: '/fseqfilelist' saved 76 bytes.
[11:19:58][FileMgr] Configuration File: '/config.json' loaded.
[11:19:58][FileMgr] Configuration File: '/input_config.json' loaded.
[11:19:58]Stop server!
[11:19:59][FileMgr] Configuration File: '/output_config.json' loaded.
[11:20:00][FileMgr] Configuration File: '/admininfo.json' saved 172 bytes.
[11:20:00] Ftp server waiting for connection on port 21
[11:20:00][FileMgr] Configuration File: '/config.json' saved 530 bytes.
[11:20:00][WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
[11:20:00][WiFiDrv] Using DHCP
[11:20:00][WiFiDrv] Connecting to 'paddycakes-2.4g' as Octa2Go
[11:20:06][WiFiDrv] WiFi Entering State: Connected To AP
[11:20:06][WiFiDrv] Connected with IP: 192.168.1.181
[11:20:06][WebMgr] Web server listening on port 80
[11:20:06]Stop server!
[11:20:06][  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[11:20:06] Ftp server waiting for connection on port 21
[11:20:06][  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320

MartinMueller2003, 2025-06-30:
I updated my dig octa and it has been running without issue playing a file from the SD for 2 days. I do see a crash caused by OOM when xLights does a fpp discovery. Still trying to find a way around that.