forkineye/ESPixelStick #857: Hard crash when accessing web pages

jeffu231, 2025-07-10:
Running on clean build from main branch built using espsv3 environment in VSCode under PlatformIO. https://github.com/forkineye/ESPixelStick/commit/5ccb320552d9a8782e917800d94323f92d1f4dc5

Martins head branch exhibits the same behavior.
https://github.com/MartinMueller2003/ESPixelStick/commit/0ca75201dd37d8ce83a0a1309a442898970109ad

Using Shelbys ESPPixelstick with D1 Mini and blank SD card inserted.

Serial monitor connected on hardware usb port.

Primary input set to e1.31
Secondary input set to FPP Remote
Output set to WS2811

Everything else left to defaults. No changes to defaults for the above options.

Wifi connected to normal AP.

After configuring inputs and outputs switching back to the Home tab  it experiences a crash with the following stack trace.

Browser console shows calls for 
/conf/admininfo.json 200 Ok
/conf/config.json 200 Ok
/XJ   failed to connect. So this seems like the suspect api call driving the crash.

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4024506d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40285df0 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe940 end: 3fffffb0 offset: 0150
3fffea90:  40285df0 00000004 3fff0000 40285df0  
3fffeaa0:  00000004 3fff0000 3fff7ed4 0000007c  
3fffeab0:  40285df0 00000004 3fff0000 3ffe864c  
3fffeac0:  40285df0 3fff7ec4 3fffec10 40207994  
3fffead0:  3fff7ed4 0000007c 3fff7ed4 0000ffff  
3fffeae0:  3fff7ed4 3fff7ec4 3fffec10 4020fc04  
3fffeaf0:  40285df0 00000004 3fff0000 3fff7ed4  
3fffeb00:  0000007c 3fffed54 0000007b 40208000  
3fffeb10:  40285df0 00000004 3fff0000 00000000  
3fffeb20:  00000004 3fff7ecc 3fff7ebc 3fff0079  
3fffeb30:  0000000b 00000000 3fff2500 3fff7ebc  
3fffeb40:  3fff1850 3fffec10 3fff2570 40214044  
3fffeb50:  3fff7ebc 3fff7ec4 3fffec10 40285df0  
3fffeb60:  00000004 3fff0000 00000005 4020fee8  
3fffeb70:  3fff7ec4 3fff007a 3fff7ec4 3fff007a  
3fffeb80:  40286738 00000005 3fff7ebc 3fff7ebc  
3fffeb90:  3fff1850 3fffec10 3fffec10 3fff7ebc  
3fffeba0:  3fff1850 3fffec10 3fff7ec4 3fff7ebc  
3fffebb0:  3fff1850 3fffec10 3fff2570 4021fd24  
3fffebc0:  3fff7ec4 3fffec10 3fffec10 40286738  
3fffebd0:  00000005 3fff7e8c 3fffec10 65736e75  
3fffebe0:  00000000 00000000 000b000f 00000000  
3fffebf0:  3fff7dac 3fffec10 3fffec10 00000000  
3fffec00:  402852f8 3fffec10 00003d58 4021083c  
3fffec10:  3ffe864c 00000000 3fff85e4 007d0080  
3fffec20:  3fff7af4 00000040 00000020 401017ac  
3fffec30:  3fff5944 00000040 3fff7bc8 3fffec1c  
3fffec40:  00040001 0000ffff 00010000 ffff0020  
3fffec50:  402852f8 3fff7b6c 3fffec10 3fff7f94  
3fffec60:  0000000d 00000000 0000001f 40100cc8  
3fffec70:  ffffffff 00000000 3fffc228 401067d9  
3fffec80:  00000000 000d000f 00000000 00000001  
3fffec90:  4000e0d8 00000030 00000018 ffffffff  
3fffeca0:  40241764 3fff782c 3fff77a4 00000045  
3fffecb0:  20302e35 00000003 3fff783c 3fff77b4  
3fffecc0:  00000007 614d206c 63614d28 3fff7960  
3fffecd0:  00000075 3fff77a4 00000083 00000030  
[... 291 lines cut ...]
3fffff10:  00000000 00012b9c 6083126e 015a6ccf  
3fffff20:  3fff4854 4022ede8 3fff2714 3fff3fd4  
3fffff30:  40243fc8 3fff27c0 3fff27c0 402058bc  
3fffff40:  00000000 3fff3e18 3fff3fa8 4023f970  
3fffff50:  3fffdad0 00000000 3fff3fa8 40222b0b  
3fffff60:  00000000 00000000 00000000 0032003f  
3fffff70:  00000000 00000000 002d002f 3fffff5c  
3fffff80:  00000000 0017ffff 00010000 ffff0020  
3fffff90:  0015001f 00000000 00000000 3fff3fd4  
3fffffa0:  3fffdad0 00000000 3fff3fa8 3fff3fd4  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00092bd0
~ld

MartinMueller2003, 2025-07-16:
Latest version on my branch has some changes that mitigate this. It comes at the price of loading a bit slower.

jeffu231, 2025-07-16:
I pulled the latest from your branch and I am still seeing the hard crash. Clean build of firmware and file system. Same steps to reproduce.

Let me know if there is anything else I can provide to help.

Commit ab7f8e23044c5507e4b2552b5d1292b382bec732

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40245a55 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402867f8 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe940 end: 3fffffb0 offset: 0150
3fffea90:  402867f8 00000004 3fff0000 402867f8  
3fffeaa0:  00000004 3fff0000 3fff90c4 0000007c  
3fffeab0:  402867f8 00000004 3fff0000 3ffe864c  
3fffeac0:  402867f8 3fff90b4 3fffec10 402076cc  
3fffead0:  3fff90c4 0000007c 3fff90c4 0000ffff  
3fffeae0:  3fff90c4 3fff90b4 3fffec10 4020fc38  
3fffeaf0:  402867f8 00000004 3fff0000 3fff90c4  
3fffeb00:  0000007c 3fffed54 0000007b 40207d3c  
3fffeb10:  402867f8 00000004 3fff0000 00000000  
3fffeb20:  00000004 3fff90bc 3fff90ac 3fff0079  
3fffeb30:  0000000b 00000000 3fff2c00 3fff90ac  
3fffeb40:  3fff1910 3fffec10 3fff2c90 402140ec  
3fffeb50:  3fff90ac 3fff90b4 3fffec10 402867f8  
3fffeb60:  00000004 3fff0000 00000005 4020ff70  
3fffeb70:  3fff90b4 3fff007a 3fff90b4 3fff007a  
3fffeb80:  40287180 00000005 3fff90ac 3fff90ac  
3fffeb90:  3fff1910 3fffec10 3fffec10 3fff90ac  
3fffeba0:  3fff1910 3fffec10 3fff90b4 3fff90ac  
3fffebb0:  3fff1910 3fffec10 3fff2c90 4021ff70  
3fffebc0:  3fff90b4 3fffec10 3fffec10 40287180  
3fffebd0:  00000005 3fff907c 3fffec10 65736e75  
3fffebe0:  00000000 00000000 000b000f 00000000  
3fffebf0:  3fff8f9c 3fffec10 3fffec10 00000000  
3fffec00:  40285d00 3fffec10 00002ec0 402108b8  
3fffec10:  3ffe864c 3ffe9700 3fff9804 007d0080  
3fffec20:  3fff8ce4 3fffec30 3fff8944 402396e8  
3fffec30:  696c6e69 0000656e 3fff8958 3fffec1c  
3fffec40:  00040001 3fffffff 00010000 ffff0020  
3fffec50:  40285d00 3fff8d5c 3fffec10 3fff9184  
3fffec60:  0000000d 00000000 4027b917 00000001  
3fffec70:  ffffffff 00000000 3ffeb861 00000008  
3fffec80:  00000000 000d000f 00000000 00000001  
3fffec90:  00000002 00000000 00000020 40100ce0  
3fffeca0:  00000005 00000005 00000002 40102cc0  
3fffecb0:  00000005 00000000 00000020 40100ce0  
3fffecc0:  00000000 40263c8b 00000005 40102cc0  
3fffecd0:  3ffeb715 40105f57 3ffef128 3ffeee58  
3fffece0:  4010384b 3ffef128 3fff61e0 0000000d  
3fffecf0:  0000000d 00000010 00000020 401017c0  
3fffed00:  0000000d 00000010 3fffedc8 40242046  
3fffed10:  00000000 00000010 3fffedd4 40242046  
3fffed20:  00000000 0000052c 3ffe8718 3fff8d3c  
3fffed30:  3fffec10 3fff8d2c 3fffec10 3fff8d1c  
3fffed40:  3fffec10 3fff8d5c 3fffec10 3fff8cfc  
3fffed50:  3fffec10 3fff8cec 3fffec10 3fffec48  
3fffed60:  3fffec10 00000001 3fff6398 40234e64  
3fffed70:  3fff86e4 00002f70 00000000 00000018  
3fffed80:  00000000 0000000a 3fff8cd0 00000002  
3fffed90:  00000002 3fff8a19 3fff86e4 4023527c  
3fffeda0:  3fff86e4 3fff608c 3fff86e4 3fff15f0  
3fffedb0:  00000002 3fff8a19 3fff86e4 40236fa8  
3fffedc0:  3fff86e4 00000001 40247304 402472f8  
3fffedd0:  00000002 3fff8a19 3fff86e4 402370f4  
3fffede0:  3fff86e4 00000458 40247318 40236f88  
3fffedf0:  00000002 3fff8a19 3fff86e4 40237226  
[... 273 lines cut ...]
3fffff10:  00000000 00009675 07ae147a 3fff487c  
3fffff20:  3fffdad0 3fff3660 3fff3160 4022660a  
3fffff30:  3fff3160 3fff4680 40246b14 40227208  
3fffff40:  3fffdad0 3fff3560 3fff3160 402266b9  
3fffff50:  3fffdad0 00000000 3fff4850 40222dbf  
3fffff60:  00000000 00000000 00000000 0038003f  
3fffff70:  00000000 00000000 0033003f 3fffff5c  
3fffff80:  00000000 0019ffff 00010000 ffff0020  
3fffff90:  00000000 000c000f 00000000 3fff487c  
3fffffa0:  3fffdad0 00000000 3fff4850 3fff487c  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000936c0
~ld

jeffu231, 2025-07-16:
Looks like it may have clipped the first few lines as the monitor connected after the reset.

[device-monitor-250716-142008.log.zip](https://github.com/user-attachments/files/21264576/device-monitor-250716-142008.log.zip)

jeffu231, 2025-07-16:
Here is another one that looks like a full capture. 

[device-monitor-250716-143407.log.zip](https://github.com/user-attachments/files/21264786/device-monitor-250716-143407.log.zip)

jeffu231, 2025-07-18:
The latest on your branch has solved the crashes I was experiencing. I am doing some Vixen dev work to incorporate uploading FSEQ to these, so I will report any other issues I come across.