forkineye/ESPixelStick #860: Device reboots after uploading fseq file

jeffu231, 2025-07-18:
ESPixelStick from Shelby ESP8266 D1 Mini

After uploading a v2 fseq file via POST request to /fpp?path=uploadFile&filename=AmazingGrace-Techno.fseq the device does a reboot. 

Code was from head of Martins branch: 66af98615d99740f1da5df79c75f486c0958ecf8

If a reboot is only required if it is a xlz extension, this code seems to be evaluating that incorrectly.

From FileMgr.cpp line 2071. I suspect the temp.indexOf(".xlz") should be temp.endsWith(".xlz") since indexOf will return -1 when not found and evaluates to true. I can submit a PR if this is the desired behavior or you can incorporate it yourself if that is easier.

String temp = String(fsUploadFileName);
        temp.toLowerCase();
        if(temp.indexOf(".xlz"))
        {
            String reason = F("Reboot after receiving a compressed file");
            RequestReboot(reason, 100000);
        }

[  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[FileMgr] Configuration File: '/fseqfilelist.json' saved 85 bytes.
[FileMgr] Upload File: 'AmazingGrace-Techno.fseq' Started
[FileMgr] Configuration File: '/fseqfilelist.json' saved 85 bytes.
␛[Fprogress: 1277, heap: 15072
␛[Fprogress: 2737, heap: 15048
␛[Fprogress: 3913, heap: 15280
.
.
.
.
␛[Fprogress: 1475593, heap: 14232
␛[Fprogress: 1477053, heap: 14232
␛[Fprogress: 1478513, heap: 14208
␛[Fprogress: 1479064, heap: 15024
[FileMgr] Upload File: 'AmazingGrace-Techno.fseq' Done (84s). Received: 1479064 Bytes out of 1479064 bytes. FileLen: 1479064
[FileMgr] Configuration File: '/fseqfilelist.json' saved 161 bytes.
[   ESP] ***-----Internal Reboot Requested: 'Reboot after receiving a compressed file' Rebooting Now
[WiFiDrv] WiFi Lost the connection to the AP
[WiFiDrv] WiFi Entering State: Connection **Failed**

jeffu231, 2025-07-20:
I can confirm that your main repo corrects the issue in my simple upload test case. Also tested a simple zipped version of the same file that had the extension of .xlz and the device rebooted after receiving.

patdelaney, 2025-07-20:
@MartinMueller2003  can you get this into the webflasher?

MartinMueller2003, 2025-07-20:
As soon as I fix an issue with the JS terser function.

MartinMueller2003, 2025-07-21:
This is now in Experimental on the web flash tool.

MartinMueller2003, 2025-07-21:
This fix has been merged into main.

closed 2025-07-21 by MartinMueller2003