forkineye/ESPixelStick #867: e1.31 Configuration limits universe start to values <= 255

jeffu231, 2025-08-04:
Universe numbers can be higher than 255. Suggest the max field limit should be much higher like 65535 to allow for more flexibility with larger displays.

File html/e1_31.html line 6 where the max input value is constrained.

MartinMueller2003, 2025-08-04:
Experimental version as of now has this change.

jeffu231, 2025-08-04:
In theory someone may want to just use a less channels than fully supported in the last few universes, so it should really be bounded by the number of max universes -  universes actually used by channel count. But that is probably a very corner case and a lot more effort for such a case. This certainly opens it up for broader use and solves for my use case. 

Thanks

MartinMueller2003, 2025-09-18:
This has been added to main